yaqwsx/PcbDraw-Lib #1: Documentation for the creation of new component

kwendenarmo, 2017-04-28:
I love this project, I am currently doing manual documentation of boards and this greatly simplifies the process.

The PcbDraw script works amazingly! But when creating new components I get a lot of problems:

-   What is the SVG default unit? In the documentation you says "one default unit corresponds to 1 mm in reality.", but in the SVGs of the library I opened, none of the red rectangles measure 1x1.
-   The red rectangle representing the origin, where should the origin be located? That is, which of its corners or center represents the coordinate 0,0 {X, Y}?
-   The size of the canvas of the SVG file, does it fulfill some function and has to have a specific size, or on the contrary can be depending on the content? In all built-in SVGs of the library, the canvas is much larger than the component.
-   Is there any color-coding convention to use? I mean standard colors to represent the body of an IC, the metal of a pin, etc.
-   Do the new components that we create have to be placed following some specific path? As can be the same path as the kicad module (library_name/module_name).

I use Inkscape as a native tool for editing SVG files.

yaqwsx, 2017-04-28:
Hi, thank you for the feedback.
- I have no deeper understanding of the SVG file format and its relation to graphical programs. I created all my modules with Inkscape. PcbDraw opens the SVG file takes out all the elements an treats its size as milimiters (this is due to the restriction of paths, which cannot specify units)
- Origin is defined by XY the position of SVG element with id "origin". For a rectangle, this is a top-left corner
- Canvas is not taken in account
- So far, I haven't came up with a convention. So far, I took average color from a board photo.
- All the new modules has to follow the path of corresponding KiCAD footprint. E.g. module for a footprint "MyIC" from a library "MyLib" has to be placed in KiCAD-base/MyLib/MyIC.svg

kwendenarmo, 2017-04-29:
Okay, I think I understand how it works. Then I have another question: Does the origin of the component in SVG have to match the anchor of the module in KiCad? I understand that the point of origin must be the same to correctly position the image of the component.

Thank you very much for the answers, I will try to create some soon.

yaqwsx, 2017-04-29:
Yes, the origin has to be at the same position as the KiCAD footprint anchor.

kwendenarmo, 2017-05-12:
I have tried to create a simple component (0805 SMD resistor) but I have not been able to make PcbDraw show it on the board.

I have been analyzing one of the components provided (0603 SMD resistor), and I have not found any difference, beyond size, shape and color. I have seen that the red rectangle has transparency. This is necessary?

I have also seen that the default unit is px, instead of millimeter. Since I had drawn it in millimeters, it is much larger than it should have been. I understand that this is not a problem for PcbDraw (that is, it is not the reason why it does not show it).

PcbDraw finds the SVG file in the library (does not display the message "Warning: component 'R_0805_HandSoldering' from library 'Resistors_SMD' was not found"). I have tried to create a link to the SMD resistor 0603 already created and show it correctly, so the bug has to be in my file. I have tried to save the file as Inkscape SVG (default) or flat SVG, no results in any case.

Any idea what might be going on?
Thanks.

yaqwsx, 2017-05-12:
Can you attach your SVG file?

kwendenarmo, 2017-05-12:
I can not attach SVG files here, but it is in this link https://github.com/akornsys-rdi/SVG-library-and-palette/blob/master/R_0805.svg

yaqwsx, 2017-05-29:
Hi, sorry for a late response. I have been quite busy recently. I looked at your SVG and the reason for it to not work correctly is that you origin element is under transformation as it is in a `g` element:

```
inkscape:label="Capa 1"
inkscape:groupmode="layer"
id="layer1"
transform="translate(-158.24236,-671.53617)"
```

If I put it outside this group, it works. I admit it would be nice to take the transformation in account, however there are higher priority tasks I need to handle first. If you have time, PR would be appreciated!

One small note: I think your 0805 resistor is oversized (compare it to you origin element, which is 1x1 mm).

yaqwsx, 2017-06-06:
@kwendenarmo Does it solve your issue?

yaqwsx, 2018-07-25:
I implemented support for SVG transformations in PcbDraw. Feel free to test it!

yaqwsx, 2021-05-21:
Making the components should be now fully straightforward due to 06b8c2cb8f62beffdd93d6f78c3223cd94d1b860.

closed 2021-05-21 by yaqwsx