yaqwsx/PcbDraw-Lib #2: Submissions for components outside of standard KiCad libs

NT7S, 2017-05-29:
Excellent project, thank you very much for it!

I'm more than happy to start submitting PRs for components that I create, but I have a question on how you would like to handle components that aren't in the standard libs. A fair bit of the footprints that I use are in my own custom internal library, but I'm sure they could be useful for others. Since the directory name in this lib has to match the KiCad footprint library directory name, it wouldn't be appropriate for me to submit my custom components under the folder where they usually live.

Any thoughts on how you would like to handle this?

yaqwsx, 2017-05-29:
If the components can useful to the others (and I think this is nearly every component except custom boards/modules), please place them in the KiCAD-base directory (as I have already started with my own library yaqwsx_KiCAD_lib). I think the names of the libraries and modules will be general enough and there won't be any collisions.

NT7S, 2017-05-29:
OK, so leaving them in my internal library folder is good? Will do.

Should see a PR pretty soon. Not a ton of components, but some more to work with.

yaqwsx, 2017-05-29:
Perfect, can I close this issue?

NT7S, 2017-05-29:
Yes, thanks for the clarification

closed 2017-05-29 by NT7S