yaqwsx/PcbDraw-Lib #8: Distribute the libraries

FFY00, 2020-07-11:
Hi, I am packaging pcbdraw for Arch Linux and was wondering if maybe the libraries could be packaged too. We could distribute the libraries and have pcbdraw include them by default.

A simple makefile that puts the libraries in `/usr/share/pcbdraw` would work (just make sure you respect `prefix` and `DESTDIR`). This could also be packaged via setuptools, I think.

Then pcbdraw would just look for the files there and include them :grin:.

yaqwsx, 2020-07-12:
This is related to https://github.com/yaqwsx/PcbDraw/pull/31 and https://github.com/yaqwsx/PcbDraw/pull/29

I plan to review these PRs and merge them in a few days.

yaqwsx, 2020-08-05:
Just an update - both PRs were merged.

Salamandar, 2020-10-22:
@FFY00 A simple "cp" or "mv" in the package() function of the PKGBUILD would work, right ? :)

yaqwsx, 2021-05-21:
What's the situation on this?

Salamandar, 2021-05-21:
@yaqwsx So let me summarize.
* If no lib is passed as argument (or kicad-default) footdraws are searched at /usr/share/pcbdraw ?
* If a lib is passed, is /usr/share/pcbdraw searched or only the path passed as argument