yaqwsx/PcbDraw-Lib #17: create template doesn't work

picli3, 2021-12-14:
sorry to write at all times. make the change you indicated and I already have my PCB with the capacitors and resistors, however I try to create a new template to create a new model and I get this error
[image]

I want to make all the components of this PCB, I hope you can help me.

[image]

yaqwsx, 2021-12-14:
What does `create_template.py --help` say? It is weird that the option is not recognized - it is in the source code https://github.com/yaqwsx/PcbDraw-Lib/blob/b0a42a1cddd7441ea4b743c8ce922e911f9f2b36/scripts/create_template.py#L117

picli3, 2021-12-16:
this is the message i get

[image]

yaqwsx, 2021-12-16:
If you read the help, you will the problem - the script has two commands: one for building a template for every component on the board and one for building it out of a single footprint. And you did not specify this command. You simply miss the word `footprint`.

picli3, 2021-12-16:
With this command it does not throw an error, but when I see the output it is only a red square.

`scripts/create_template.py footprint /usr/share/kicad/modules/Package_BGA.pretty/LFBGA-100_10x10mm_Layout10x10_P0.8mm.kicad_mod test/so2.svg`

[image]

yaqwsx, 2021-12-16:
Could you share the footprint and resulting SVG?

picli3, 2021-12-16:
this is the output file.

thanks for your help.

[so2.svg](https://drive.google.com/file/d/1SLy3_p9K0YaLX4YOgWtlTdPfO5N0mKM2/view?usp=sharing)

yaqwsx, 2021-12-16:
[image]

It seems that you have created a template for the bottom view of the component. Please, try specifying the `--front` switch explicitly.

picli3, 2021-12-16:
> sorry to write at all times. make the change you indicated and I already have my PCB with the capacitors and resistors, however I try to create a new template to create a new model and I get this error [image]

if specific that I get this error.

yaqwsx, 2021-12-16:
You again miss the command `footprint`. You have to specify **both** `footprint` and `--front`. May I ask what lead you to the conclusion that you can omit the `footprint` part of the command?

picli3, 2021-12-16:
you have it like this in the documentation, I don't see the `footprint` command 

`scripts/create_template.py --front \
  /usr/share/kicad/modules/Resistor_SMD.pretty/R_1206_3216Metric.kicad_mod \
  KiCAD-base/Resistor_SMD.pretty/R_1206_3216Metric.svg`

yaqwsx, 2021-12-16:
Thanks for pointing that out. That is a mistake; the proper command is:
```
scripts/create_template.py footprint --front path/to/footprint.kicad_mod path/to/template.svg
```

picli3, 2021-12-16:
I will try it when I get home. thank you!

picli3, 2021-12-17:
now it worked, thank you very much

`scripts/create_template.py footprint --front /usr/share/kicad/modules/Package_BGA.pretty/LFBGA-100_10x10mm_Layout10x10_P0.8mm.kicad_mod test/so2.svg`

[image]

closed 2021-12-17 by picli3

yaqwsx, 2021-12-17:
For reference, the command was fixed in 47bd6d71dde1aa4563655a3bc03fac0eeb3de854