yaqwsx/PcbDraw-Lib #20: TypeError: get_layers() takes 3 positional arguments but 4 were given

dev-techshlok, 2021-12-21:
I am getting this error and after correcting it in create_template.py file i am getting new error that is TypeError: File or filename expected, got 'LazyFile'
[image]
[image]
.

yaqwsx, 2021-12-21:
What is the version of PcbDraw you have installed? The most recent version of `create_template.py` definitely requires the most recent version from Git (though, this is not documented yet).

Please, for future reference, attach the stack traces as text, not a screenshot. It allows me to copy-paste from them. Thanks!

dev-techshlok, 2021-12-21:
Hii
Thankyou for your response.
Version : 

PcbDraw 0.8.0

Error: 

Traceback (most recent call last):
  File "/home/Downloads/PcbDraw-Lib-master/scripts/create_template.py", line 211, in <module>
    cli()
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/home/Downloads/PcbDraw-Lib-master/scripts/create_template.py", line 135, in run_footprint
    document.getroot().append(plotTopLayers(board))
  File "/home/techshlok/Downloads/PcbDraw-Lib-master/scripts/create_template.py", line 71, in plotTopLayers
    elements = pcbdraw.get_layers(board, colors, {},plotPlan)
TypeError: get_layers() takes 3 positional arguments but 4 were given

Error after removing one argument from pcbdraw.py file.

Traceback (most recent call last):
  File "/home/Downloads/PcbDraw-Lib-master/scripts/create_template.py", line 211, in <module>
    cli()
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/home/Downloads/PcbDraw-Lib-master/scripts/create_template.py", line 142, in run_footprint
    document.write(output)
  File "src/lxml/etree.pyx", line 2057, in lxml.etree._ElementTree.write
  File "src/lxml/serializer.pxi", line 745, in lxml.etree._tofilelike
  File "src/lxml/serializer.pxi", line 812, in lxml.etree._create_output_buffer
  File "src/lxml/serializer.pxi", line 808, in lxml.etree._create_output_buffer
TypeError: File or filename expected, got 'LazyFile'

yaqwsx, 2021-12-21:
As I said, this (v0.8) is not the latest (upstream) version from GitHub: see https://github.com/yaqwsx/PcbDraw/blob/master/doc/installation.md

dev-techshlok, 2021-12-21:
hii

after Installing latest pcbdraw vesrion when I was checking the version it gave me this error:
Traceback (most recent call last):

ImportError: cannot import name 'KICAD_VERSION' from 'pcbnewTransition' (/home/.local/lib/python3.8/site-packages/pcbnewTransition/__init__.py)

yaqwsx, 2021-12-21:
You also have to bump pcbnewTransition to v0.2. (the stuff between proper releases can be messy).

dev-techshlok, 2021-12-21:
HI
It is already showing V0.2.

What exactly I have to do ?