yaqwsx/PcbDraw-Lib #24: Global Lookup Table

Electro707, 2022-01-24:
Kind of related to https://github.com/yaqwsx/PcbDraw/issues/66, and the change will ultimately need to be done in pcbdraw, but I am creating this Issue here as it relates to this repository.

Instead of symlinking footprint names, wouldn't it be better to implement some kind of lookup table per footprint library that pcbdraw will read and link component. Symlinks eventually become quite cluttered, especially with footprint renames and supporting older versions of KiCAD's footprint library:
[image]

yaqwsx, 2022-02-15:
What library table/format do you propose? 

I think a better solution would be to preserve the symlinks but change the library structure:
```
libraries/
├── PcbDraw/
│   ├── Resistors
│   ├── Capacitors
│   └── ...
├── KiCADv5/
│   ├── Resistors.pretty
│   └── ...
└── KiCADv6/
     ├── Resistors.pretty
     └── ...
```

Where `PcbDraw` library is the only library containing actual images and all the other libraries are only collections of symlinks. What do you think?

Electro707, 2022-02-23:
I was originally thinking just a json file with the format
```json
  "KICAD_NAME1": "SVG_PATH1",
```
The main reason I am suggesting a single lookup file over symlinks is that it allows easier changes over time and having one centralized location to know what drawing a footprint is referencing to. 

The symlink structure you suggested I also like, and will hopefully clean up and separate the drawings folder so it's not cluttered. 
I am personally indifferent to either method.

n-i-x, 2023-05-20:
Another argument in favor of the footprint remapping suggestion is for making this library work with EasyEDA designs.

I'm able to generate SVG from KiCad PCB converted with https://wokwi.com/tools/easyeda2kicad which uses https://github.com/wokwi/easyeda2kicad. But all of the converted footprints have library names like `easyeda:R0402`. I've been working around this by creating symlinks in an easyeda library to KiCad footprints, but it would be pretty nice if I could just have an eadyeda map file I can pass in and it just performs magic.