yaqwsx/PcbDraw-Lib #25: module 'pcbnew' has no attribute 'PCB_IO'

Sampozzo, 2022-02-28:
Hello,

i'm trying to convert a footprint with the convert script but i have this error.
Using Kicad 6.0 on Ubuntu. Can you help me please?

`./scripts/create_template.py footprint --front ../Libs/ZioFootPrints.pretty/screw_terminal_12pins_2_54.kicad_mod ../Libs/ZioPcbDrawLib/screw_terminal_12pins_2_54.svg
Traceback (most recent call last):
  File "./scripts/create_template.py", line 210, in <module>
    cli()
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.8/dist-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "./scripts/create_template.py", line 127, in run_footprint
    board = buildFootprintBoard(footprint)
  File "./scripts/create_template.py", line 22, in buildFootprintBoard
    footprint = loadFootprint(footprintPath)
  File "./scripts/create_template.py", line 15, in loadFootprint
    return pcbnew.PCB_IO().FootprintLoad(lib, foot)
AttributeError: module 'pcbnew' has no attribute 'PCB_IO'`

yaqwsx, 2022-03-01:
The script uses v5 API that was changed for v6; `PCB_IO` was decomposed and renamed. `pcbnew.FootprintLoad` should be used instead. I will update the script soon.

Sampozzo, 2022-03-01:
@yaqwsx thank you very much, it works now :)